tpaviot/pythonocc-generator #5: forward declaration of 'BRepAdaptor_Surface'

jf---, 2015-07-29:
affected modules: _BRepClass3d.so, BRepApprox.so
possible: _ChFi3d

tpaviot, 2015-07-30:
Please paste the complete compiler trace.

jf---, 2015-07-30:
see #1

jf---, 2015-07-30:
same thing, in another module:

[  9%] Building CXX object CMakeFiles/_BRepApprox.dir/Darwin/x86_64-MinSizeRel-64/BRepApproxPYTHON_wrap.cxx.o
In file included from /Users/jelleferinga/GIT/pythonocc-core-occ-690/build/Darwin/x86_64-MinSizeRel-64/BRepApproxPYTHON_wrap.cxx:3915:
In file included from /Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_SurfaceTool.hxx:149:
/Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_SurfaceTool.lxx:31:108: error: member access into incomplete type
      'const BRepAdaptor_Surface'
inline Standard_Real BRepApprox_SurfaceTool::FirstUParameter(const BRepAdaptor_Surface& Surf){  return Surf.FirstUParameter(); }
                                                                                                           ^
/Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_Approx.hxx:23:7: note: forward declaration of 'BRepAdaptor_Surface'
class BRepAdaptor_Surface;
      ^
In file included from /Users/jelleferinga/GIT/pythonocc-core-occ-690/build/Darwin/x86_64-MinSizeRel-64/BRepApproxPYTHON_wrap.cxx:3915:
In file included from /Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_SurfaceTool.hxx:149:
/Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_SurfaceTool.lxx:32:108: error: member access into incomplete type
      'const BRepAdaptor_Surface'
inline Standard_Real BRepApprox_SurfaceTool::FirstVParameter(const BRepAdaptor_Surface& Surf){  return Surf.FirstVParameter();}
                                                                                                           ^
/Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_Approx.hxx:23:7: note: forward declaration of 'BRepAdaptor_Surface'
class BRepAdaptor_Surface;
      ^
In file included from /Users/jelleferinga/GIT/pythonocc-core-occ-690/build/Darwin/x86_64-MinSizeRel-64/BRepApproxPYTHON_wrap.cxx:3915:
In file included from /Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_SurfaceTool.hxx:149:
/Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_SurfaceTool.lxx:33:107: error: member access into incomplete type
      'const BRepAdaptor_Surface'
inline Standard_Real BRepApprox_SurfaceTool::LastUParameter(const BRepAdaptor_Surface& Surf){  return Surf.LastUParameter();}
                                                                                                          ^
/Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_Approx.hxx:23:7: note: forward declaration of 'BRepAdaptor_Surface'
class BRepAdaptor_Surface;
      ^
In file included from /Users/jelleferinga/GIT/pythonocc-core-occ-690/build/Darwin/x86_64-MinSizeRel-64/BRepApproxPYTHON_wrap.cxx:3915:
In file included from /Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_SurfaceTool.hxx:149:
/Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_SurfaceTool.lxx:34:107: error: member access into incomplete type
      'const BRepAdaptor_Surface'
inline Standard_Real BRepApprox_SurfaceTool::LastVParameter(const BRepAdaptor_Surface& Surf){  return Surf.LastVParameter();}
                                                                                                          ^
/Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_Approx.hxx:23:7: note: forward declaration of 'BRepAdaptor_Surface'
class BRepAdaptor_Surface;
      ^
In file included from /Users/jelleferinga/GIT/pythonocc-core-occ-690/build/Darwin/x86_64-MinSizeRel-64/BRepApproxPYTHON_wrap.cxx:3915:
In file included from /Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_SurfaceTool.hxx:149:
/Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_SurfaceTool.lxx:38:14: error: member access into incomplete type
      'const BRepAdaptor_Surface'
  return Surf.NbUIntervals(S);
             ^
/Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_Approx.hxx:23:7: note: forward declaration of 'BRepAdaptor_Surface'
class BRepAdaptor_Surface;
      ^
In file included from /Users/jelleferinga/GIT/pythonocc-core-occ-690/build/Darwin/x86_64-MinSizeRel-64/BRepApproxPYTHON_wrap.cxx:3915:
In file included from /Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_SurfaceTool.hxx:149:
/Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_SurfaceTool.lxx:43:14: error: member access into incomplete type
      'const BRepAdaptor_Surface'
  return Surf.NbVIntervals(S);
             ^
/Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_Approx.hxx:23:7: note: forward declaration of 'BRepAdaptor_Surface'
class BRepAdaptor_Surface;
      ^
In file included from /Users/jelleferinga/GIT/pythonocc-core-occ-690/build/Darwin/x86_64-MinSizeRel-64/BRepApproxPYTHON_wrap.cxx:3915:
In file included from /Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_SurfaceTool.hxx:149:
/Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_SurfaceTool.lxx:49:7: error: member access into incomplete type
      'const BRepAdaptor_Surface'
  Surf.UIntervals(Tab,S);
      ^
/Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_Approx.hxx:23:7: note: forward declaration of 'BRepAdaptor_Surface'
class BRepAdaptor_Surface;
      ^
In file included from /Users/jelleferinga/GIT/pythonocc-core-occ-690/build/Darwin/x86_64-MinSizeRel-64/BRepApproxPYTHON_wrap.cxx:3915:
[... cut ...]
In file included from /Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_SurfaceTool.hxx:149:
/Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_SurfaceTool.lxx:128:4: error: member access into incomplete type
      'const BRepAdaptor_Surface'
  S.D1(U,V,P,D1U,D1V);
   ^
/Users/jelleferinga/miniconda/envs/oce_0.17/include/oce/BRepApprox_Approx.hxx:23:7: note: forward declaration of 'BRepAdaptor_Surface'
class BRepAdaptor_Surface;
      ^
fatal error: too many errors emitted, stopping now [-ferror-limit=]

closed 2015-12-09 by jf---