tpaviot/pythonocc-generator #9: multiprocessing support

jf---, 2015-07-30:
multiprocessing.Pool would helps to speed up generating the SWIG files linearly

jf---, 2015-07-30:
speed up is pretty much linear

tpaviot, 2015-07-30:
The previous generator, based on gccxml, was very long to achieve the generation of swig files (more than one hour AFAIR). Multiprocessing was really useful to speed up generation. Since the move to CppHeaderParser, it's much quicker, around 5mn on a single core usual machine. I did not port the multiprocessing feature because it makes the debugging task a bit more difficult since log messages are mixed together. False by default is a must, I'm not sure I'll ever switch it to True.

jf---, 2015-07-30:
I'm not advocating for it to run in parallel by default, its makes tracing much more difficult and renders the stdout / stderr output pretty useless. Nice to have though

commit 0f03d25a (2015-08-12): Merge pull request #11 from jf---/jf/multiprocessing

fixed #9, support for multiprocessing
commit b3e56711 (2015-08-12): fixed #9, support for multiprocessing
closed 2015-08-12 by tpaviot