tpaviot/pythonocc-generator #13: running generate_wrapper.py on OCE-0.16.1

sven-q, 2015-09-08:
Hi, I'm trying to understand the pythonocc-generator and I ran it's release 0.2 on the OCE release 0.16.1 and had some problems with it:

1. I noticed that XSControl.i is missing (It may relate to issue OCE 0.17 XSControl.i missing #2 ?).

2. I replaced the SWIG files in pythonocc-core release 0.16.1 and tried to compile, it fails compiling the module `ChFiKPart` from the toolkit `TKFillet`:
```
[ 26%] Building CXX object CMakeFiles/_ChFiKPart.dir/src/SWIG_files/wrapper/ChFiKPartPYTHON_wrap.cxx.o
In file included from /local/bzfherrm/test/build/pythonocc/src/SWIG_files/wrapper/ChFiKPartPYTHON_wrap.cxx:4258:0:
/local/bzfherrm/test/install/oce/lib/oce-0.16/../../include/oce/ChFiKPart_ComputeData_Rotule.hxx:20:39: error: ‘TopOpeBRepDS_DataStructure’ was not declared in this scope
 Standard_Boolean ChFiKPart_MakeRotule(TopOpeBRepDS_DataStructure& DStr,
                                       ^
[...]
```
`git diff` on the related SWIG files `ChFiKPart.i` gives:
```
diff --git a/src/SWIG_files/wrapper/ChFiKPart.i b/src/SWIG_files/wrapper/ChFiKPart.i
index 5646842..af2dcbc 100644
--- a/src/SWIG_files/wrapper/ChFiKPart.i
+++ b/src/SWIG_files/wrapper/ChFiKPart.i
@@ -2 +2 @@
-Copyright 2008-2014 Thomas Paviot (tpaviot@gmail.com)
+Copyright 2008-2015 Thomas Paviot (tpaviot@gmail.com)
@@ -44,0 +45 @@ class ChFiKPart_ComputeData {
+               %feature("compactdefaultargs") Compute;
@@ -62 +63 @@ class ChFiKPart_ComputeData {
-       :type Iedge: Standard_Integer
+       :type Iedge: int
@@ -65,0 +67 @@ class ChFiKPart_ComputeData {
+               %feature("compactdefaultargs") ComputeCorner;
@@ -98,0 +101 @@ class ChFiKPart_ComputeData {
+               %feature("compactdefaultargs") ComputeCorner;
@@ -127,0 +131 @@ class ChFiKPart_ComputeData {
+               %feature("compactdefaultargs") ComputeCorner;
@@ -172,0 +177 @@ class ChFiKPart_DataMapIteratorOfRstMap : public TCollection_BasicMapIterator {
+               %feature("compactdefaultargs") ChFiKPart_DataMapIteratorOfRstMap;
@@ -175,0 +181 @@ class ChFiKPart_DataMapIteratorOfRstMap : public TCollection_BasicMapIterator {
+               %feature("compactdefaultargs") ChFiKPart_DataMapIteratorOfRstMap;
@@ -180,0 +187 @@ class ChFiKPart_DataMapIteratorOfRstMap : public TCollection_BasicMapIterator {
+               %feature("compactdefaultargs") Initialize;
@@ -185,0 +193 @@ class ChFiKPart_DataMapIteratorOfRstMap : public TCollection_BasicMapIterator {
+               %feature("compactdefaultargs") Key;
@@ -188,0 +197 @@ class ChFiKPart_DataMapIteratorOfRstMap : public TCollection_BasicMapIterator {
+               %feature("compactdefaultargs") Value;
@@ -211,0 +221 @@ class ChFiKPart_DataMapNodeOfRstMap : public TCollection_MapNode {
+               %feature("compactdefaultargs") ChFiKPart_DataMapNodeOfRstMap;
@@ -213 +223 @@ class ChFiKPart_DataMapNodeOfRstMap : public TCollection_MapNode {
-       :type K: Standard_Integer &
+       :type K: int &
@@ -234 +244,2 @@ class ChFiKPart_DataMapNodeOfRstMap : public TCollection_MapNode {
-                       %feature("autodoc", "   :rtype: Handle_Adaptor2d_HCurve2d
+                       %feature("compactdefaultargs") Value;
+               %feature("autodoc", "   :rtype: Handle_Adaptor2d_HCurve2d
@@ -295,0 +307 @@ class ChFiKPart_RstMap : public TCollection_BasicMap {
+               %feature("compactdefaultargs") ChFiKPart_RstMap;
@@ -297 +309 @@ class ChFiKPart_RstMap : public TCollection_BasicMap {
-       :type NbBuckets: Standard_Integer
+       :type NbBuckets: int
@@ -300,0 +313 @@ class ChFiKPart_RstMap : public TCollection_BasicMap {
+               %feature("compactdefaultargs") Assign;
@@ -305,0 +319 @@ class ChFiKPart_RstMap : public TCollection_BasicMap {
+               %feature("compactdefaultargs") operator =;
@@ -310,0 +325 @@ class ChFiKPart_RstMap : public TCollection_BasicMap {
+               %feature("compactdefaultargs") ReSize;
@@ -312 +327 @@ class ChFiKPart_RstMap : public TCollection_BasicMap {
-       :type NbBuckets: Standard_Integer
+       :type NbBuckets: int
@@ -315,0 +331 @@ class ChFiKPart_RstMap : public TCollection_BasicMap {
+               %feature("compactdefaultargs") Clear;
@@ -318,0 +335 @@ class ChFiKPart_RstMap : public TCollection_BasicMap {
+               %feature("compactdefaultargs") Bind;
@@ -320 +337 @@ class ChFiKPart_RstMap : public TCollection_BasicMap {
-       :type K: Standard_Integer &
+       :type K: int &
@@ -325,0 +343 @@ class ChFiKPart_RstMap : public TCollection_BasicMap {
+               %feature("compactdefaultargs") IsBound;
@@ -327 +345 @@ class ChFiKPart_RstMap : public TCollection_BasicMap {
-       :type K: Standard_Integer &
+       :type K: int &
@@ -330,0 +349 @@ class ChFiKPart_RstMap : public TCollection_BasicMap {
+               %feature("compactdefaultargs") UnBind;
@@ -332 +351 @@ class ChFiKPart_RstMap : public TCollection_BasicMap {
-       :type K: Standard_Integer &
+       :type K: int &
@@ -335,0 +355 @@ class ChFiKPart_RstMap : public TCollection_BasicMap {
+               %feature("compactdefaultargs") Find;
@@ -337 +357 @@ class ChFiKPart_RstMap : public TCollection_BasicMap {
-       :type K: Standard_Integer &
+       :type K: int &
@@ -340,0 +361 @@ class ChFiKPart_RstMap : public TCollection_BasicMap {
+               %feature("compactdefaultargs") ChangeFind;
@@ -342 +363 @@ class ChFiKPart_RstMap : public TCollection_BasicMap {
-       :type K: Standard_Integer &
+       :type K: int &
@@ -345,0 +367 @@ class ChFiKPart_RstMap : public TCollection_BasicMap {
+               %feature("compactdefaultargs") Find1;
@@ -347 +369 @@ class ChFiKPart_RstMap : public TCollection_BasicMap {
-       :type K: Standard_Integer &
+       :type K: int &
@@ -350,0 +373 @@ class ChFiKPart_RstMap : public TCollection_BasicMap {
+               %feature("compactdefaultargs") ChangeFind1;
@@ -352 +375 @@ class ChFiKPart_RstMap : public TCollection_BasicMap {
-       :type K: Standard_Integer &
+       :type K: int &
```
and on `ChFiKPart_headers.i` it shows that the order is different, which may cause the compiling error.

Am I using the python-generator in a wrong way? Thanks in advance for helping!
sven

sven-q, 2015-09-09:
2. Ok, sorry, the first differences were my fault, I took the wrong pythonocc release, now with 0.16.2 the `ChFiKPart.i` is the same.
But `ChFiKPart_headers.i` is still produced with a different order of the includes and compiling leads to the same error quoted above (`[...] error: ‘TopOpeBRepDS_DataStructure’ was not declared in this scope [...]`). Here a detaiil from the `git diff`:
```
diff --git a/src/SWIG_files/wrapper/ChFiKPart_headers.i b/src/SWIG_files/wrapper/ChFiKPart_headers.i
index fc987fa..0026a71 100644
--- a/src/SWIG_files/wrapper/ChFiKPart_headers.i
+++ b/src/SWIG_files/wrapper/ChFiKPart_headers.i
@@ -18,1404 +18,1404 @@ along with pythonOCC.  If not, see <http://www.gnu.org/licenses/>.
 
 */
 %{
-#include<ChFiKPart_ComputeData.hxx>
-#include<ChFiKPart_ComputeData_CS.hxx>
-#include<ChFiKPart_ComputeData_Fcts.hxx>
-#include<ChFiKPart_ComputeData_FilPlnCon.hxx>
-#include<ChFiKPart_ComputeData_FilPlnCyl.hxx>
-#include<ChFiKPart_ComputeData_FilPlnPln.hxx>
 #include<ChFiKPart_ComputeData_Rotule.hxx>
+#include<ChFiKPart_ComputeData_FilPlnCyl.hxx>
 #include<ChFiKPart_DataMapIteratorOfRstMap.hxx>
 #include<ChFiKPart_DataMapNodeOfRstMap.hxx>
+#include<ChFiKPart_ComputeData_FilPlnCon.hxx>
 #include<ChFiKPart_RstMap.hxx>
+#include<ChFiKPart_ComputeData.hxx>
+#include<ChFiKPart_ComputeData_CS.hxx>
+#include<ChFiKPart_ComputeData_Fcts.hxx>
+#include<ChFiKPart_ComputeData_FilPlnPln.hxx>
 #include<TopOpeBRepDS.hxx>
-#include<TopOpeBRepDS_Array1OfDataMapOfIntegerListOfInterference.hxx>
-#include<TopOpeBRepDS_Association.hxx>
-#include<TopOpeBRepDS_BuildTool.hxx>
-#include<TopOpeBRepDS_Check.hxx>
-#include<TopOpeBRepDS_CheckStatus.hxx>
-#include<TopOpeBRepDS_Config.hxx>
-#include<TopOpeBRepDS_connex.hxx>
-#include<TopOpeBRepDS_Curve.hxx>
-#include<TopOpeBRepDS_CurveData.hxx>
-#include<TopOpeBRepDS_CurveExplorer.hxx>
-#include<TopOpeBRepDS_CurveIterator.hxx>
-#include<TopOpeBRepDS_CurvePointInterference.hxx>
-#include<TopOpeBRepDS_DataMapIteratorOfDataMapOfCheckStatus.hxx>
-#include<TopOpeBRepDS_DataMapIteratorOfDataMapOfIntegerListOfInterference.hxx>
-#include<TopOpeBRepDS_DataMapIteratorOfDataMapOfInterferenceListOfInterference.hxx>
-#include<TopOpeBRepDS_DataMapIteratorOfDataMapOfInterferenceShape.hxx>
-#include<TopOpeBRepDS_DataMapIteratorOfDataMapOfShapeListOfShapeOn1State.hxx>
-#include<TopOpeBRepDS_DataMapIteratorOfDataMapOfShapeState.hxx>
-#include<TopOpeBRepDS_DataMapIteratorOfMapOfCurve.hxx>
-#include<TopOpeBRepDS_DataMapIteratorOfMapOfIntegerShapeData.hxx>
-#include<TopOpeBRepDS_DataMapIteratorOfMapOfPoint.hxx>
-#include<TopOpeBRepDS_DataMapIteratorOfMapOfSurface.hxx>
 #include<TopOpeBRepDS_DataMapIteratorOfShapeSurface.hxx>
-#include<TopOpeBRepDS_DataMapNodeOfDataMapOfCheckStatus.hxx>
-#include<TopOpeBRepDS_DataMapNodeOfDataMapOfIntegerListOfInterference.hxx>
-#include<TopOpeBRepDS_DataMapNodeOfDataMapOfInterferenceListOfInterference.hxx>
-#include<TopOpeBRepDS_DataMapNodeOfDataMapOfInterferenceShape.hxx>
 #include<TopOpeBRepDS_DataMapNodeOfDataMapOfShapeListOfShapeOn1State.hxx>
-#include<TopOpeBRepDS_DataMapNodeOfDataMapOfShapeState.hxx>
-#include<TopOpeBRepDS_DataMapNodeOfMapOfCurve.hxx>
-#include<TopOpeBRepDS_DataMapNodeOfMapOfIntegerShapeData.hxx>
-#include<TopOpeBRepDS_DataMapNodeOfMapOfPoint.hxx>
-#include<TopOpeBRepDS_DataMapNodeOfMapOfSurface.hxx>
-#include<TopOpeBRepDS_DataMapNodeOfShapeSurface.hxx>
-#include<TopOpeBRepDS_DataMapOfCheckStatus.hxx>
-#include<TopOpeBRepDS_DataMapOfIntegerListOfInterference.hxx>
-#include<TopOpeBRepDS_DataMapOfInterferenceListOfInterference.hxx>
-#include<TopOpeBRepDS_DataMapOfInterferenceShape.hxx>
-#include<TopOpeBRepDS_DataMapOfShapeListOfShapeOn1State.hxx>
-#include<TopOpeBRepDS_DataMapOfShapeState.hxx>
-#include<TopOpeBRepDS_DataStructure.hxx>   <-------
-#include<TopOpeBRepDS_define.hxx>
[... 72 lines cut ...]
+#include<TopOpeBRepDS_DataMapNodeOfMapOfIntegerShapeData.hxx>
+#include<TopOpeBRepDS_CurveIterator.hxx>
+#include<TopOpeBRepDS_BuildTool.hxx>
+#include<TopOpeBRepDS_Curve.hxx>
+#include<TopOpeBRepDS_repvg.hxx>
+#include<TopOpeBRepDS_Config.hxx>
+#include<TopOpeBRepDS_DataMapOfCheckStatus.hxx>
+#include<TopOpeBRepDS_Surface.hxx>
+#include<TopOpeBRepDS_DataMapNodeOfMapOfPoint.hxx>
+#include<TopOpeBRepDS_DoubleMapNodeOfDoubleMapOfIntegerShape.hxx>
 #include<TopOpeBRepDS_ListIteratorOfListOfInterference.hxx>
-#include<TopOpeBRepDS_ListNodeOfListOfInterference.hxx>
-#include<TopOpeBRepDS_ListOfInterference.hxx>
+#include<TopOpeBRepDS_DoubleMapOfIntegerShape.hxx>
+#include<TopOpeBRepDS_DataMapIteratorOfDataMapOfInterferenceShape.hxx>
+#include<TopOpeBRepDS_DataMapNodeOfShapeSurface.hxx>
 #include<TopOpeBRepDS_ListOfShapeOn1State.hxx>
+#include<TopOpeBRepDS_CurveExplorer.hxx>
+#include<TopOpeBRepDS_EXPORT.hxx>
+#include<TopOpeBRepDS_Interference.hxx>
+#include<TopOpeBRepDS_DataStructure.hxx>   <-------
[...]
```
I ran the `generate_wrapper.py` on linux with python 2.7.6.
How can I fix the order of the includes? Thanks!

sven-hm, 2015-09-21:
Ok, I could fix the include order using pythons list sort, see [here](https://github.com/sven-hm/pythonocc-generator/tree/sven-hm/fix_include_order).

But I don't understand why you get "well sorted" `*_headers.i` files?

closed 2016-10-11 by tpaviot