tpaviot/pythonocc-generator #17: generate type hints for python >= 3.5

jf---, 2015-12-14:
[see pep 484](https://www.python.org/dev/peps/pep-0484/#stub-files)

python can accept stub files that describe the type ( in / output ) of a function or method.
this is incredible useful, since IDE's such as pycharm will be able to check ~80/90% of your code, rather than ~40/50% as of today. also, python linters can work with the type checking to test the code for correctness.

It would be great for the code generator to produce type stub files ( *.pyi ). This way, the same python wrappers can be used whether python supports type hinting or not ( eg. if python_version >= 3.5 then copy the stub files )

Certainly when working with a larger code base, this is a huge help to develop robust software rapidly.
Typed languages have an important advantage when refactoring a large code base, you can do so much faster with more confidence.

Personally, this is an important motivation to move to py3k on short notice :sunglasses:

lalvarezt, 2016-10-18:
@jf--- Hi, I've been working on something similar. Check the code here [https://github.com/lalvarezt/pythonocc-generator/commit/cce82eb4b32cb4cb178681cd819e7e0e6311a920](url).

This generates the additional .pyi files that you referred to. It's not complete yet, I need to manage the overload cases, for now it only adds the first occurrence of an overload function. I checked gp and graphic3d module and it looks OK. I'm generating all the modules now. If you have the time check to see if the generated code is OK, I've tested it with PyCharm and it works for the selected modules.

jf---, 2016-10-18:
oh my @lalvarezt , that's super exciting 🚀 

what about the more explicit way as how [mypy](http://mypy.readthedocs.io/en/latest/python2.html) handles type hinting for python 2?

perhaps having inlined type hints is preferable to .pyi files, since its valuable documentation / avoids having yet another file opened in your editor...

lalvarezt, 2016-10-18:
@jf--- I've tried that already, the problem I found is that on the .i files you can only affect the GetObject and GetHandle functions, since they are explicitly on the code. Another thing is that swig generates the typing information using the c/c++ type and then writes it on the python generated file. So from what I saw the procedure is the following
-Edit .i
-Compile
-Edit .py to correct the type

Its much easier to generate the .pyi files, and copy them. Actually you would never have to open the .pyi because the only information on them is the type.

lalvarezt, 2016-10-18:
6e7fede28114c466ba5428242cb15ebefd02570d

closed 2023-07-30 by tpaviot