tpaviot/pythonocc-generator #22: use libclang to parse headers

jf---, 2016-01-30:
CppHeaderParser cannot parse fairly big list of file.
Therefore, it might be worthwhile to consider using [libclang's](http://szelei.me/code-generator/) python bindings as a parser.

tpaviot, 2016-01-30:
It took me two years to move from gccxml to cppheaderparser. Is it a joke ?

closed 2016-01-30 by tpaviot

jf---, 2016-01-30:
I dont mean to step on your toes Thomas...

But glancing at the code, __parse_header__ is called in a single place.
If another parser generates the same data ( while using another parser backend ) as the CppHeaderParser.CppHeader method than that's OK, right?

here the code in context:

```python
    cpp_headers = map(parse_header, module_headers)
    module_typedefs = {}
    module_enums = []
    module_classes = {}
    module_free_functions = []
    for header in cpp_headers:
        # builde the typedef dictionnary
        module_typedefs = dict(module_typedefs.items() + header.typedefs.items())
        # build the enum list
        module_enums += header.enums
        # build the class dictionary
        module_classes = dict(module_classes.items() + header.classes.items())
        # build the free functions list
        module_free_functions += header.functions
```

Perhaps a bit of background; spend most of the week re-implementing [Font_BRepFont](https://github.com/tpaviot/oce/blob/master/src/Font/Font_BRepFont.cxx) ( i forgot that this was implemented in OCE, its even on a list "interesting classes to further explore" ). I think this is one of the ( few ) classes that isnt wrapped pythonocc-core. Hence, my interest to wrap that class. Also, I think learning about the mechanics of the project is relevant and interesting.

Again, please dont take offence....