tpaviot/pythonocc-generator #28: Automatically downcasting base class pointers with SWIG

alynch4047, 2016-10-15:
Problem Statement:

When getting the selected objects from the view, they are returned as a list of TopoDS_Shape. However, the objects that are returned are usually actually of a derived class from TopoDS_Shape such as TopoDS_Face or TopoDS_Solid. This is a nuisance when wanting to pass to methods that demand a specific subclass such as BRepTool_Surface which wants a TopDS_Face and not a TopDS_Shape. In this case I could not find a straightforward way of telling python that my TopoDS_Shape was in fact a TopoDS_Face. It is possible is to tell SWIG (see below ) that there is a class hierarchy with its base at TopoDS_Shape, and that when it is returning a TopDS_Shape, SWIG can then look at e.g. shape.ShapeType() to see what class of python wrapper to put the returned object into, rather than just wrapping it as a base TopoDS_Shape.

The benefit of this is that the python user will receive their objects wrapped in the correct derived class and not a base class for that genera of objects.

Proposed Solution:

This can be achieved using SWIG's typemap (out) capability: see the two blog posts below

http://stackoverflow.com/questions/27392602/swig-downcasting-from-base-to-derived

http://johnnado.com/swig-csharp-java-downcast/

rainman110, 2018-11-24:
You can use swigs factory feature, see
https://github.com/DLR-SC/tigl/blob/c62f4e0ebbc3ed9accbf46a2a51f21421ce590ed/bindings/python_internal/configuration.i#L406

tpaviot, 2018-11-28:
Fix suggested at https://github.com/tpaviot/pythonocc-core/pull/587

tpaviot, 2018-11-28:
Fixed.

closed 2018-11-28 by tpaviot

alynch4047, 2018-11-30:
Sorry about the late reply, this is great thanks Thomas