tpaviot/pythonocc-generator #36: Upgrade in OCE to 6.9.1

lalvarezt, 2016-11-08:
@tpaviot A few days ago there was an update on the OCE repo, upgrading OCCT to version 6.9.1. I compiled the code and it looks fine. I'm trying to update the generator, but there are some (a lot) of compile problems, I'm trying to fix them, if possible. When I have something new I'll let you know.

lalvarezt, 2016-11-09:
@tpaviot I updated the code and after some small changes, pythonocc-core compiles, I tested the examples and it looks normal, but I'm having some performance issues with the new library, don't know why. With the new library issue #139 of pythonocc-core is fixed. I'll post some pictures on the other repo.

jf---, 2016-11-09:
@lalvarezt , so you've managed to generate wrapper for 6.9.1? wow, that does seem very promising...

lalvarezt, 2016-11-10:
@tpaviot I created a new PR that fixes some things that are necessary for wrapping the current master. Of course, the PR also fixes problems in the previous master.

lalvarezt, 2016-11-11:
@tpaviot I think I've got everything ready to generate the wrap, but first is necessary to apply a patch to OCE, after that the current generator with the latest updates should work without problems.

tpaviot, 2016-11-14:
@lalvarezt We use to adapt the python wrapper to oce, not the opposite. Indeed modifying oce may lead to incompatibilities with upstream. Remember that user should be able to switch from oce back/to occ as they want. We must ensure binary compatibility with oce and occ.

jf---, 2016-11-14:
a stopgap measure can be that for the conda can also take a patch file
perhaps this is a way to move forward, while formal changes to OCE are made?

lalvarezt, 2016-11-14:
@tpaviot @jf--- I understand what you are saying, but the changes on the OCE repo are problems that needed to be fixed IMO. If you look at the changes you'll notice that there's even code that was missing for a function, that I had to add it from OCCT. For example the function ErrorStatus was missing in the BRepLib_CheckCurveOnSurface.lxx file.

jf---, 2016-11-14:
well, OCE 0.18 ( -> OCCT 6.9.1 ) is not released yet, officially... 
why not be pragmatic, and create corresponding OCE branch, such to be able to move forward?
@lalvarezt your efforts are much appreciated and its good to have a better understanding to grasp where the friction occurs, so that's interesting...

when you mention that the `ErrorStatus` method was missing, is your diff from OCCT 6.9.1 -> OCE 0.18? 

( just to be sure, i think @tpaviot's remark wrt binary compatibility is key indeed )

lalvarezt, 2016-11-14:
@jf--- @tpaviot  Yes ErrorStatus is missing in OCE 6.9.1 not in OCCT. I looked up the code and added it. I agree, binary compatibility is important, the changes are minor and don't interfere with that IMO.

jf---, 2016-11-14:
@tpaviot , [this is the branch](https://github.com/lalvarezt/oce/tree/Minor_patches) of OCE being discussed

jf---, 2016-11-22:
Gents,

What is a logical next step?

- release OCE 0.18?
- @lalvarezt , you've built pythonocc-core with OCE 0.18, can you inform us to what extent tests / examples are passing?
- with OCE 0.18 released ( or at least a conda build ) at least we could see the status on TravisCI, which is important to advance the project
- I think it would be wise to release a 0.17.1 on short notice, considerable improvements have been made...

trelau, 2017-01-23:
Does the master branch include any updates for wrapping OCE 0.18.0? I'm trying to get myself proficient with the pythonoocc build process and test out some surface intersection improvements in OCCT 6.9.1/OCE 0.18.0. I get a handful of the modules to compile with no problem, can get more to compile if I exclude some class methods (BRepBuilderAPI_FastSewing :: GetStatuses and BRepExtrema_OverlapTool :: OverlapTriangles). I also had to grab BOPCol_Array1.hxx from OCCT 6.8.0 and drop that into the OCE header folder. But even then I get a whole bunch of syntax errors. I think about 25 projects fail to build.

I'm mostly curious if the syntax errors are something that need updated in the current master or if I'm doing something wrong on my end.

jf---, 2017-01-23:
hi @trelau , great initiative... so just to be clear; are you wrapping OCE 0.18 or OCCT 6.9.1?
that's pretty relevant info...

trelau, 2017-01-23:
OCE 0.18.0..the pre-compiled version for Win64 MSVC2015

trelau, 2017-02-27:
I was able to wrap OCE 0.18.0 after applying the patches [referenced above](https://github.com/lalvarezt/oce/tree/Minor_patches). I had to apply my own shown as screen shots below because I modified the headers of the pre-compiled versions for now..

Modify the Box() method
[image]

Include Select3D_SensitiveSet
[image]

After some tweaks to the pythonocc-generator...I am able to compile pythonocc-core 0.18.0. The only issues seem to be when I launch the viewing tools...

[image]

Seems like the Standard GLSL programs are hard-coded from the conda build...can anyone point me in the right direction for a fix? Thanks.

tpaviot, 2017-02-28:
The GLSL path is hardcoded in oce, certainly. I have to take a look at this. In the meantime, you can setup the CSF_ShadersDirectory environment variable to point to the right location.

tpaviot, 2017-05-07:
Fixed by PR #42

closed 2017-05-07 by tpaviot