tpaviot/pythonocc-generator #39: Template Wrapping

lalvarezt, 2016-11-14:
@tpaviot Currently template use in the OCE is a problem when wrapping with SWIG. On the current master this problem is minor. This could be a major problem when moving to OCCT 7 or higher, due to the new type refactoring, almost all classes are represented as templates.

From the doc:

"In order to wrap a template, you need to give SWIG information about a particular template instantiation (e.g., vector<int>, array<double>, etc.). Second, an instantiation name such as vector<int> is generally not a valid identifier name in most target languages. Thus, you will need to give the template instantiation a more suitable name such as intvector when creating a wrapper."

After declaring the template you need to instantiate it (for example for int and double):

template<class T> class List {
...
};

%template(intList) List<int>;
%template(doubleList) List<double>;

This means that is necessary to know before hand what are the possible cases for each template in order to generate the necessary wrapper code.

I have a possible workaround. To perform an static analysis on the code in order to "detect" the possible representations for the template.

Any thoughts?

jf---, 2016-11-14:
I've worked with ITK which is a image processing library wrapped by swig, which uses templates extensively. It might provide an interesting analogy?

in the old days, the project used gccxml for parsing and has recently moved to it successor [castxml](https://www.youtube.com/watch?v=O2lBgtaDdyk). itk's wrapper generation code is [here](https://github.com/InsightSoftwareConsortium/ITK/tree/master/Wrapping/Generators) 

perhaps this can provide as reference to manage templates?

when parsing the C++ code, its possible to get the list of arguments, that is the types that are parameters for the template?

tpaviot, 2016-11-14:
@lalvarezt I fully agree with you. Moving up the wrapper to the coming oce-0.18 (==occt691) is straight forward, there will be some minor issues that will easily be solved.

In a midterm, I'm a lot more concerned about the latest occt7 version. I'm aware of the major refactoring about the handles, with a intensive use of c++ templates. So far, I did not reallly dive into the wrapping of templates, but hey, I guess this moment gets closer and closer ;-)

To be honest, I did not work enough to have a clear idea about how things should be done. I don't even have a clue about the different solutions that could be investigated. Please @lalvarezt go on with this idea of a pre-static analysis to detect templates. In a first STEP, there's no need to modify the generator, we can work on the pythonocc-core repository, with manual changes to some .i files. When we clearly understand what happens, it will be time for an automation of the wrapper generation.

tpaviot, 2016-11-14:
@jf--- thanks for introducing castxml, I was not aware of that project. And ITK is certainly a good place to grab good ideas with SWIG and c++

lalvarezt, 2016-11-14:
@jf--- Any help is appreciated.

jf---, 2016-11-22:
@lalvarezt , is there news on handling templates?
Perhaps we should focus on some design / experimental implementations to advance this topic.
Possible @rainman110 has an interesting outlook on the topic?

tpaviot, 2016-12-20:
@lalvarezt I started looking at the template issue. I just pushed the latest CppHeaderParser release (previous one had an error handling templates).

Templated class methods should don't be that hard to wrap.

Not the same for templated classes (for instance NCollection_*), it seems to be a bit more difficult.

IMO we should go step by step. The first one is to wrap templated methods with a predefined and default "list" of accepted types, e.g. float, int, bool, str. Additionnal types can be specified in the Modules.py script.

I'm about to open a PR related to this, I will notify you as soon as it is ready for review.

lalvarezt, 2016-12-20:
@tpaviot That's very good news, I was working with the latest CppHeaderParser release, that's why I didn't notice the error. Adding specialized types in the Modules.py it's a great idea.

tpaviot, 2019-02-01:
See Issue #49

closed 2019-02-01 by tpaviot