tpaviot/pythonocc-generator #40: OCCT Voxel package

aothms, 2017-02-09:
Hi,

I grepped through Modules.py and it appears the Voxel package [1] is not wrapped. Just checking in whether there is a reason for this? I think manipulating this kind of data with `numpy.ndarray`s would be quite a thrill :)

[1] https://www.opencascade.com/doc/occt-6.9.1/refman/html/package_voxel.html

Kind regards,
Thomas

tpaviot, 2017-02-10:
@aothms I don't remember why it's not wrapped ;) Maybe because I failed to... I will try to investigate this

jf---, 2017-02-10:
the numpy angle is interesting...
numpy would be great for working with meshes...

aothms, 2017-02-12:
Well, it turns out that (read) access to the underlying data structure is not really feasible/possible anyway. Calling `Voxel_DS::Get()` for every voxel is not likely going to be performant. In the end I wrote a small C++ app that voxelises the geometry. And a small Python code to parse the output of `Voxel_Writer`. Not a high priority issue imho.

jf---, 2017-02-13:
@aothms, sounds like a perfect case for numpy?

aothms, 2017-02-14:
@jf---  Definitely, the output of `Voxel_Writer` is not a dump that numpy immediately understands, it is some sort of custom sparse format. But I have my voxels in a `numpy.ndarray(dtype=numpy.bool_)` now and it works like a charm :)

jf---, 2017-02-14:
@aothms so there's a /contrib folder in pyocc which is a useful location for those rare cases where its more straightfwd to wrap a C++ implementation. If you feel like contributing this, that's probably the right place

aothms, 2017-02-27:
Ok, so it seems that the Voxel package was removed altogether from OCCT 7 [1] [2]. This in favour of their ``Volume rendering'' apparently. I didn't come across this, but perhaps they refer to their proprietary offerings [3]. Personally I think it's a shame, I mean, voxels are suitable for things beyond visualisation, like analyses that are simply not viable on a topological brep model. But anyway all the more reason to close this issue.

[1] https://www.opencascade.com/sites/default/files/documents/Release_Notes_7.0.0.pdf
[2] http://git.dev.opencascade.org/gitweb/?p=occt.git;a=commit;h=6ce0df1e78b1554048cd4b08346aa3166b4c55b0
[3] https://www.opencascade.com/content/volume-rendering-and-computed-tomography-support

closed 2017-02-27 by aothms