tpaviot/pythonocc-generator #49: Work In Progress: Expose OCCT 7.2.0

tpaviot, 2019-02-01:
Follow Issue #30. Related to https://github.com/tpaviot/pythonocc-core/issues/576

Issue dealing with the process of wrapping occt 7.2.0 using SWIG and expose related objects through pythonocc.

I'm currently being porting the pythonocc wrapper to occt 7.2.0 version. This requires many changes to both SWIG files (automatically generated using the pythonocc-generator project), and to most of the pythonocc python scripts impacted by API changes.

Changes to the generator are available in this repository, branch https://github.com/tpaviot/pythonocc-generator/tree/wip/occt7x-wrapper . In this branch, support for SMESH and Addon has been temporarily removed. They should be back as soon as the basis is working.

I ran the generator over all occt 7.2.0 header files, resulting SWIG files are available on the branch wip/occt7x-support https://github.com/tpaviot/pythonocc-core/tree/wip/occt7x-support SWIG files are here https://github.com/tpaviot/pythonocc-core/tree/wip/occt7x-support/src/SWIG_files/wrapper

The related pythonocc-core branch should compile, on Linux, using occt7.2.0. I had to add a few missing includes, see https://github.com/tpaviot/oce/commits/review/occt-720-patches

After a ```$ make install```, I can run the unittest suite. Currently pass, see the output below:
```
pythonocc-core/test$ python run_tests.py 
testAutoImportOfDependentModules (core_wrapper_features_unittest.TestWrapperFeatures)
Test: automatic import of dependent modules ... ok
testProtectedConstructor (core_wrapper_features_unittest.TestWrapperFeatures)
Test: protected constructor ... ok
testStandardBooleanByRefPassedReturned (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
testTopoDS_byref_arguments (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_array_iterator (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_const_Standard_Integer_byref (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_const_Standard_Real_byref (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_default_constructor_DEFINE_STANDARD_ALLOC (core_wrapper_features_unittest.TestWrapperFeatures)
OCE classes the defines standard alllocator can be instanciated ... ok
test_deprecation_downcasts (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_deprecation_get_handle (core_wrapper_features_unittest.TestWrapperFeatures)
Handles are now completely transparent. The GetHandle method is ... ok
test_deprecation_get_object (core_wrapper_features_unittest.TestWrapperFeatures)
Handles are now completely transparent. The GetObject method is ... ok
test_deprecation_handle_class (core_wrapper_features_unittest.TestWrapperFeatures)
Handles are now completely transparent. The Handle_* constructor is ... ok
test_deprecation_warning (core_wrapper_features_unittest.TestWrapperFeatures)
since pythonocc-0.18.2. import OCC.* changed to import OCC.Core.* ... ok
test_dict (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_downcast_curve (core_wrapper_features_unittest.TestWrapperFeatures)
Test if a GeomCurve can be DownCasted to a GeomLine ... ERROR
test_dump_to_string (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_eq_operator (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_ft1 (core_wrapper_features_unittest.TestWrapperFeatures)
Test: Standard_Integer & by reference transformator ... ok
test_gp_Quaternion (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_gp_Vec_operators (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_handle_standard_transient_copy (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_handling_exceptions (core_wrapper_features_unittest.TestWrapperFeatures)
asserts that handling of OCC exceptions is handled correctly in pythonocc ... ok
test_hash (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_hash_eq_operator (core_wrapper_features_unittest.TestWrapperFeatures)
test that the == wrapper is ok ... ok
test_in_place_operators (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_inherit_topods_shape (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_list (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_local_properties (core_wrapper_features_unittest.TestWrapperFeatures)
Get and modify class local properties ... ok
test_memory_handle_getobject (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_neq_operator (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_pickle_from_file (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_repr_for_null_topods_shapes (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_repr_overload (core_wrapper_features_unittest.TestWrapperFeatures)
Test if repr string is properly returned ... ok
test_return_enum (core_wrapper_features_unittest.TestWrapperFeatures)
Check that returned enums are properly handled, wether they're returned ... FAIL
test_shape_conversion_as_py_none (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_standard_boolean_byref (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_standard_integer_by_ref_passed_returned (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_static_method (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
test_sub_class (core_wrapper_features_unittest.TestWrapperFeatures)
Test: subclass ... ok
test_topology (core_wrapper_features_unittest.TestWrapperFeatures) ... ok
[... 152 lines cut ...]
    self.assertEqual(los1.First(), BRepCheck_Multiple3DCurve)
AssertionError: <Swig Object of type 'BRepCheck_Status *' at 0x7fa61c8dfe40> != 5

======================================================================
FAIL: test_vertex_edge (core_extend_topology_unittest.TestExtendTopology)
----------------------------------------------------------------------
Traceback (most recent call last):
  File "/home/thomas/Devel/pythonocc-core/test/core_extend_topology_unittest.py", line 126, in test_vertex_edge
    self.assertEqual(len(edges_from_vert), topo.number_of_edges_from_vertex(vert))
AssertionError: 3 != 6

----------------------------------------------------------------------
Ran 90 tests in 2.171s

FAILED (failures=2, errors=3)
```

The most important change from 6.9.x to 7.x is an intensive use of c++ templates. OCCT left a way to ensure backward compatibility, that's why despite this huge API refactoring most of the pythonocc-core unittest amazingly pass without any major change.

However @rainman110 I need your help. I think that the OccHandle.i file you contribued needs to be updated. Indeed it's not possible to ```DownCast``` anymore. For instance, in the unittest suite, the line ```Geom_BSplineCurve.DownCast```  raises an exception. It's the same each time I try to DownCast something.

Ping @jf--- @adam-urbanczyk @trelau for information

A screenshot including an AIS_Manipulator, available from occt 7.x series
[image]

trelau, 2019-02-01:
awesome work. moving forward do you intend to keep OCE up-to-date or are you planning on just using the OCCT releases?

jf---, 2019-02-01:
> A screenshot including an AIS_Manipulator

WOW! That's wonderful! So eager to delve into the generalized fused goodness!

rainman110, 2019-02-01:
Wow... Great progress! Congratulations for this achievement.

@tpaviot I'll have a look at the handle wrapper and try to fix downcasting. I suppose, we just need some small adaptations.

tpaviot, 2019-02-02:
@trelau I plan to keep OCE up to date

Thank you @rainman110 maybe you should read the migration guide that contains information about handles  https://dev.opencascade.org/doc/overview/html/occt_dev_guides__upgrade.html#upgrade_700

rainman110, 2019-02-02:
@tpaviot I think I know, why the downcasting is not working anymore.

Under the hood, swig manages all references to objects via their handles. When downcasting one object to another type, we were exploiting the inheritance chain of the handles. For example Handle(Geom_BSplineCurve) is inherited from Handle(Geom_Curve). Swig uses the inheritance type information to check, what handle types are allowed for the downcasting arguments.

In OCCT 7, handles are standalone classes that don't use inheritance. Still, Geom_BSplineCurve is inherited from Geom_Curve, but their handles aren't. This is the reason, why swig does not accept anything else for downcasting other than Handle_Standard_Transients!

We need to make swig aware of the underlying inheritance chain of the non-handle objects. I suppose, we ca use the typecheck typemaps of swig somehow.

rainman110, 2019-02-04:
I think a narrowed down the problem and have a potential fix.

@tpaviot I cannot completely build you branch 7.x. There are still a couple of errors ( i am building on windows). So, can you already build it completely with your generated files?

tpaviot, 2019-02-05:
@rainman110 I only tested on Linux so far, still did not try on Windows. Usually, Windows comes with its bunch of compilation errors, and I have to remove classes/methods that fails from the Modules.py file. There are much less differences between Linux and osx

tpaviot, 2019-02-06:
@rainman110 No, it's this line:
```
typedef opencascade::handle <Standard_Transient> Handle_Standard_Transient;
```
If I remove it, everything goes right. It appeared after a change to generate_wrapper.py

I really have difficulties to work with SWIG warnings. They're not documented, not easy to know what to do to avoid them.

rainman110, 2019-02-06:
@tpaviot Really? Everything works.

Then, windows is still an issue. Here we have they problem, that Handle(Something) is a different class than Handle_Something and swig won't convert between them right now. This is a visual c only thing and intended by the occt devs (see standard_handle.hxx). I am still looking into this.

tpaviot, 2019-02-06:
ok I see, indeed  ```Handle(Class)``` and Handle_* are different for msvc

jf---, 2019-02-06:
> I really have difficulties to work with SWIG warnings. They're not documented, not easy to know what to do to avoid them

Would it be straightfwd that the warnings bubble up depending on a environment variable?

adam-urbanczyk, 2019-02-08:
@tpaviot  great work! Can't wait to try it out.

tpaviot, 2019-02-12:
@rainman110 I updated both the generator and the wrapper, in a simpler way. There's no need to modify the OccHandle.i script anymore, and Downcasting works as expected (on Linux)

rainman110, 2019-02-12:
@tpaviot Great! Well done. 

I'll see, what I can do for Windows. Unfortunately, It might take a while until I find the time.

rainman110, 2019-02-13:
@tpaviot Can you please commit the changed code to pythonocc-core to branch wip/occt7x-support?

tpaviot, 2019-02-13:
@rainman110 I already did it. I use to push -force to this branch. fetch your local copy and reset the wip/occtx-support to origin/wip/occt7x-support to have latest changes.

rainman110, 2019-02-13:
@tpaviot Ah okay... did not see that!

Trophime, 2019-07-04:
Any update on this?

tpaviot, 2019-11-14:
Outdated. pythonocc-core now targets opencascade 7.4.0

closed 2019-11-14 by tpaviot