tpaviot/pythonocc-generator #50: Improvement: Modularize headers into jumbo header files

rainman110, 2019-11-01:
Currently, all headers including all dependencies are put into the swig header file. This looks eg like:
```
%{
#include<gp_XYZ.hxx>
#include<gp_XY.hxx>
.
.
#include<gp...hxx>
#include<Standard_Transient.hxx>
.
.
#include<Standard_Atomic.hxx>
%}
```

This is not very modular. Assume, on of the gp_XXX includes have to be changed, it has to be done in all *_header.i, which include the gp module.

My proposal is, to put all headers of one module in one jumbo header. For the gp module, we create a header gp_module.hxx:
```c++
#pragma once
#include<gp_XYZ.hxx>
#include<gp_XY.hxx>
.
.
#include<gp_XXX.hxx>
```

Then, the gp_headers.i file just looks like:
```swig
%{
#include<gp_module.hxx>
#include<Standard_module.hxx>
%}
```

Theoretically, this is also be used together with PCH to speed up compilation.

tpaviot, 2019-11-02:
This is a good idea. This will result in smaller interface files, and at some point make the *_headers.i useless. The PCH argument is convincing as well.

rainman110, 2019-11-03:
@tpaviot If you like, I'll work on it and then make a PR.

rainman110, 2019-11-05:
@tpaviot Here's a first implementation. Everything compiles. Obviously, another include_directory has to be put into the CMakeLists.txt to make everything compile.

https://github.com/rainman110/pythonocc-generator/commit/1fe0111d2ae28428a75301dcdbb2efe69be6143b

tpaviot, 2019-11-08:
@rainman110 I'm not sure gl2ps is still necessary. I noticed the all Graphic3D_EF_PDF or Graphic_3d_EF_PS have been removed, I think gl2ps support is dropped in 7.4 see https://dev.opencascade.org/index.php?q=node/1209

tpaviot, 2019-11-08:
```grep 'gl2ps' -r . *``` does not give any result, whether it is in ```/src``` or ```/include``` folders.

rainman110, 2019-11-08:
Good catch. I'll remove it.

tpaviot, 2019-11-14:
Code was updated on the pythonocc-core side see https://github.com/tpaviot/pythonocc-core/commit/4ccfc681d992a09159d195206519695c160e2309 and https://github.com/tpaviot/pythonocc-core/commit/ada5ebcaaf4547410af64b988a702b0b5bad0ba2

rainman110, 2019-11-14:
Great 😄

tpaviot, 2019-11-14:
You were right @rainman110 , this is definitely a good solution. There's one single interface file per occ package. Now I'll try to get precompiled header work, using the cotire cmake extension or the new cmae UNITY_BUILD feature (https://cmake.org/cmake/help/latest/prop_tgt/UNITY_BUILD.html#prop_tgt:UNITY_BUILD)

commit 1c58b10c (2019-11-14): Implemented jumbo headers. For each occ module, there's now only one interface file. Fix #50
commit 33a9fca9 (2019-11-14): Implemented "jumbo" header

Fixes #50
closed 2019-11-14 by tpaviot