tpaviot/pythonocc-generator #55: Bug: Wrong position of %wrap_handle

rainman110, 2019-11-03:
In e.g. StepShape.i, some %wrap_handle(xxx) come too late. All %wrap_handle calls must be in front of the first class declaration.

tpaviot, 2019-11-04:
Interface files were updated on the pythonocc-core side (see https://github.com/tpaviot/pythonocc-core/commit/1d7982b06a1a6fc0aa05baa39a490eb317498bb3). However, there's not any difference at compilation time, what about runtime ?

rainman110, 2019-11-04:
I am working on it. It requires many new imports. Takes a while to go through it manually.

rainman110, 2019-11-04:
@tpaviot What I am currently working on is the following:

I renamed all opencascade::handle<xxx> occurences to OccHandle_XXX. This OccHandl_XXX is defined in %wrap_handle.

Therefore, to successfully compile the module, the wrap_handle of a class must be imported. This way, you can make sure to not forget any import related to handles.

rainman110, 2019-11-04:
IMHO, failing early (at compile time) is much better than failing at runtime (at some user).

It requires too many tests to check all of these problems at runtime.

rainman110, 2019-11-05:
This is not yet fixed in the generator! There are still wrap_handles for the H-Classes, e.g. %wrap_handle(StepShape_HArray1OfShapeDimensionRepresentationItem) at the bottom of StepShape.i

rainman110, 2019-11-05:
I fixed it locally. Shall I make a PR?

tpaviot, 2019-11-05:
No need for a PR, I cherry-picked it. Thank you.

commit 8eac0a4c (2019-11-04): * removed Standard_Dump.hxx header to prevent TCollection and Standard circular references issues

* switched typedef and templates definition in interface files (fixes #55)

* removed timestamp

* added missing IGESToBRep wrapper (fixes #54)
commit 45225314 (2019-11-05): Fixed wrong position of wrap_handle

Fixes #55
closed 2019-11-05 by rainman110