tpaviot/pythonocc-generator #60: Update correct module dependecies

rainman110, 2019-11-04:
I used the whole day to figure out the correct dependencies. Please find attached the following file.
[dependencies.txt](https://github.com/tpaviot/pythonocc-generator/files/3805359/dependencies.txt)

tpaviot, 2019-11-04:
Great work, very interesting. What do you exactly call a "correct dependency", i.e. how did you build this oriented graph ? Indeed, according to the definition, one can consider that Standard depends upon TCollection and TCollection depends on Standard as well, which is not the case in your document, and which create circular dependencies between python modules.

rainman110, 2019-11-04:
These are the minimum requirements how to make handles work correctly. I figured it out as written in some of the other issues. In practice, I simply added dependencies until I could compile everything.

So, if there occurs a Handle(geom...), the geom model is obviously required. 

I did not look for circular dependencies though. Also did not try get to load modules. These are the minimum requirements to make everything compile.

rainman110, 2019-11-04:
So... There are definitely some circular dependencies making problems. The question is, how to untie them.

rainman110, 2019-11-04:
The smallest cycles are:

```
import Graphic3d
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/localdata1/sigg_ma/src/pythonocc-core/buildocct740/OCC/Core/Graphic3d.py", line 156, in <module>
    import OCC.Core.Select3D
  File "/localdata1/sigg_ma/src/pythonocc-core/buildocct740/OCC/Core/Select3D.py", line 158, in <module>
    import OCC.Core.SelectMgr
  File "/localdata1/sigg_ma/src/pythonocc-core/buildocct740/OCC/Core/SelectMgr.py", line 161, in <module>
    import OCC.Core.PrsMgr
  File "/localdata1/sigg_ma/src/pythonocc-core/buildocct740/OCC/Core/PrsMgr.py", line 162, in <module>
    import OCC.Core.Prs3d
  File "/localdata1/sigg_ma/src/pythonocc-core/buildocct740/OCC/Core/Prs3d.py", line 163, in <module>
    import OCC.Core.V3d
  File "/localdata1/sigg_ma/src/pythonocc-core/buildocct740/OCC/Core/V3d.py", line 357, in <module>
    class V3d_AmbientLight(OCC.Core.Graphic3d.Graphic3d_CLight):
AttributeError: module 'OCC.Core' has no attribute 'Graphic3d'
```
and
```
StepRepr
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/localdata1/sigg_ma/src/pythonocc-core/buildocct740/OCC/Core/StepRepr.py", line 150, in <module>
    import OCC.Core.StepShape
  File "/localdata1/sigg_ma/src/pythonocc-core/buildocct740/OCC/Core/StepShape.py", line 144, in <module>
    import OCC.Core.StepGeom
  File "/localdata1/sigg_ma/src/pythonocc-core/buildocct740/OCC/Core/StepGeom.py", line 1096, in <module>
    class StepGeom_GeomRepContextAndGlobUnitAssCtxAndGlobUncertaintyAssCtx(OCC.Core.StepRepr.StepRepr_RepresentationContext):
AttributeError: module 'OCC.Core' has no attribute 'StepRepr'
```

rainman110, 2019-11-05:
- In the first case, the best way to resolve the cycle is to drop the dependency of graphic3d from select3d. there are just two functions involved.

tpaviot, 2019-11-05:
Yesterday, I tried something : removing all imports from Graphic3d (at the end of the interface file, all the import Something.i). Of course it works and break the circular references between all these packages. I will try your suggestion. If it's ok, then let's go that way.

On the midterm, if circular references become a stronger issue, it may become necessary to create a single GraphicStuff.i interface file including all of the Graphic3d, PrsMgr, Select3d etc. definitions.

tpaviot, 2019-11-05:
@rainman110 btw, what's the code to draw the graph ? I guess you've been using networkx but I didn't practice this package for a while

rainman110, 2019-11-05:
@tpaviot  No, used used the dot tool. It's standard on linux/maxOS. You can directly throw in the dependencies.txt into the tool.

rainman110, 2019-11-05:
If we remove the class ```StepRepr_ReprItemAndMeasureWithUnitAndQRI``` we solve the circular dependency as well.

tpaviot, 2019-11-05:
Tested with Graphic3d. I pushed a change to the generator and updated interfaces files as well see

rainman110, 2019-11-05:
Fixed

closed 2019-11-05 by rainman110