tpaviot/pythonocc-generator #62: Missing TopoDS_Shape constructor

rainman110, 2019-11-07:
At the moment, there is no way to create an empty TopoDS_Shape. This is however required in some cases.

Therefore, the disabled constructor has to be enabled again (see Modules.py=

tpaviot, 2019-11-07:
Right, it was removed (I don't remember why), but is widely used in the examples section. It has to come back.

rainman110, 2019-11-07:
I know already, why you removed it. It pulls in two constructors. One with a template argument, which is not put correctly into the swig input file. Hence, you'll get compilation errors.

It would be nice to be able to exactly specify, which methods to prune when there are multiple versios of one method.

tpaviot, 2019-11-07:
Currently all methods names that are excluded, there's no way to choose if we can keep some of them.

Ugly hack at 9034dc5. But that works (tested)

closed 2019-11-08 by rainman110