tpaviot/pythonocc-generator #63: OCCT 7.4.0: BRepMesh_Delaun.hxx must be first BRepMesh header file

rainman110, 2019-11-11:
Otherwise, I get a compiler error on linux. Is there a possibility to modify include order?

tpaviot, 2019-11-11:
Header includes are ordered using an alphabetical sort. Just to know, which os/compiler do you use ?

tpaviot, 2019-11-11:
and what kind of error is raised by the compiler ?

rainman110, 2019-11-11:
gcc 4.8.5. The problem is, that a header - BRepMesh_BaseMeshAlgo - uses BRepMesh_Delaun before including it. This is an OCCT issue. I get the following error:
```
Building CXX object CMakeFiles/_BRepMesh.dir/Unix/x86_64-Release-64/BRepMeshPYTHON_wrap.cxx.o
In file included from /localdata1/sigg_ma/src/pythonocc-core/buildocct740/Unix/x86_64-Release-64/BRepMeshPYTHON_wrap.cxx:3926:0:
/localdata1/sigg_ma/src/opencascade-7.4.0/build/install/include/opencascade/BRepMesh_CustomDelaunayBaseMeshAlgo.hxx: In member function ‘virtual void BRepMesh_CustomDelaunayBaseMeshAlgo<BaseAlgo>::postProcessMesh(BRepMesh_Delaun&)’:
/localdata1/sigg_ma/src/opencascade-7.4.0/build/install/include/opencascade/BRepMesh_CustomDelaunayBaseMeshAlgo.hxx:48:96: error: invalid use of incomplete type ‘class BRepMesh_DataStructureOfDelaun’
     std::pair<Standard_Integer, Standard_Integer> aCellsCount = this->getCellsCount (aStructure->NbNodes());
                                                                                                ^
In file included from /localdata1/sigg_ma/src/pythonocc-core/buildocct740/Unix/x86_64-Release-64/BRepMeshPYTHON_wrap.cxx:3916:0:
/localdata1/sigg_ma/src/opencascade-7.4.0/build/install/include/opencascade/BRepMesh_BaseMeshAlgo.hxx:25:7: error: forward declaration of ‘class BRepMesh_DataStructureOfDelaun’
 class BRepMesh_DataStructureOfDelaun;
       ^
In file included from /localdata1/sigg_ma/src/pythonocc-core/buildocct740/Unix/x86_64-Release-64/BRepMeshPYTHON_wrap.cxx:3926:0:
/localdata1/sigg_ma/src/opencascade-7.4.0/build/install/include/opencascade/BRepMesh_CustomDelaunayBaseMeshAlgo.hxx:49:14: error: invalid use of incomplete type ‘class BRepMesh_Delaun’
     theMesher.InitCirclesTool (aCellsCount.first, aCellsCount.second);
              ^
In file included from /localdata1/sigg_ma/src/pythonocc-core/buildocct740/Unix/x86_64-Release-64/BRepMeshPYTHON_wrap.cxx:3916:0:
/localdata1/sigg_ma/src/opencascade-7.4.0/build/install/include/opencascade/BRepMesh_BaseMeshAlgo.hxx:26:7: error: forward declaration of ‘class BRepMesh_Delaun’
 class BRepMesh_Delaun;
       ^
```

rainman110, 2019-11-11:
How do you write class definitions into swig files?

Here, you also have to take care for class definition order, right? This could be used to correctly define header include order as well.

tpaviot, 2019-11-11:
Class definitions are processed/written to interface file from the inheritance hierarchy. First bases classes, then classes that inherit from the previous one and so one. This is why only classes with single inheritance are taken into account.

For the BRepMesh_Delaun.hxx issue, I will manually add it to the top list of headers, it's the same for the <windows.h> header to be included in AddApp2Var see https://github.com/tpaviot/pythonocc-generator/blob/wip/occt7x-wrapper/src/generate_wrapper.py#L1951

tpaviot, 2019-11-11:
Fixed by commit a5c7748265bed85db72af897161c6b0d463c09d6

rainman110, 2019-11-11:
The same is also required for the XBRepMesh package. It also includes BRepMesh stuff in the wrong order.

tpaviot, 2019-11-11:
so what's the correct order of headers in this package ?

rainman110, 2019-11-11:
The same. BrepMesh_Delaun first.

closed 2019-11-13 by rainman110