tpaviot/pythonocc-generator #84: Bindings for OCCT v7.6.1

Krande, 2022-04-22:
Hi,

In an effort to (hopefully) help with adding support for OCCT version 7.6.1 (and also learn a bit more of the build process of pythonocc-core in general) I just wanted to share my attempts at recreating the SWIG bindings to support OCCT v7.6.1. 

## General platform and software version info:

* Attempted compilation on Windows 11
* Python 3.10 
* OCCT version:  7.6.1                h8294837_0    conda-forge
* SWIG version "4.0.2"
* MSVC compiler: 1 1916

## What I've tried

First of I modified the `wrapper_generator.conf` file with the location of my pythonocc-core fork and opencascade headers.

Now, in my initial attempt at running the `generate_wrapper.py` script failed when I ran into the following error:

```
[WARNI]      [TypeHint] Skipping type NCollection_Vector<TopoDS_Face>, seems to be a template
    self._free_functions_str, self._free_functions_pyi_str = process_methods(
  File "C:\work\code\krande-pythonocc-generator\src\generate_wrapper.py", line 2396, in process_methods
    ok_to_wrap, ok_hints = process_function(function, need_overload)
  File "C:\work\code\krande-pythonocc-generator\src\generate_wrapper.py", line 2134, in process_function
    parent_class_name = f["parent"]["name"]
TypeError: 'NoneType' object is not subscriptable
```

I traced the error back to the `RWGltf` module, and thus removed it from the `TKRWMesh` list inside the `TOOLKIT_DataExchange`. 

Second attempt at running the `generate_wrapper.py` script was then successful.

Next I tried compiling pythonocc-core after updating the CMAKELIST.txt with the appropriate version of OCCT I wished to compile for.

In my compilation attempt I ran into an error related to `Error: Template 'OSD_StreamBuffer' undefined` (excerpt below).

```
====================[ Build | all | RelWithDebInfo-VS ]=========================
"C:\Program Files\JetBrains\CLion 2021.3\bin\cmake\win\bin\cmake.exe" --build C:\work\code\krande-pythonocc-core\_build --target all
[1/910] Swig compile src/SWIG_files/wrapper/Plugin.i for python
[2/910] Swig compile src/SWIG_files/wrapper/FSD.i for python
[3/910] Swig compile src/SWIG_files/wrapper/OSD.i for python
FAILED: bin/OSDPYTHON_wrap.cxx bin/OSD.py C:/work/code/krande-pythonocc-core/_build/bin/OSDPYTHON_wrap.cxx C:/work/code/krande-pythonocc-core/_build/bin/OSD.py 
cmd.exe /C "cd /D C:\work\code\krande-pythonocc-core\_build && "C:\Program Files\JetBrains\CLion 2021.3\bin\cmake\win\bin\cmake.exe" -E make_directory C:/work/code/krande-pythonocc-core/_build/CMakeFiles/_OSD.dir bin/ bin/ && "C:\Program Files\JetBrains\CLion 2021.3\bin\cmake\win\bin\cmake.exe" -E env SWIG_LIB=C:/ProgramData/chocolatey/lib/swig/tools/install/swigwin-4.0.2/Lib C:/ProgramData/chocolatey/bin/swig.exe -python -fvirtual -py3 -w302,401,402,412,314,509,512,504,325,503,520,350,351,383,389,394,395,404 -DCSFDB -DWIN32 -D_WINDOWS -outdir bin/ -c++ -interface _OSD -IC:/work/code/krande-pythonocc-core -IC:/work/code/krande-pythonocc-core/OPENGL_INCLUDE_DIR -IC:/AibelProgs/Miniconda3/envs/pyoccdev/Library/include/opencascade -IC:/work/code/krande-pythonocc-core/src/SWIG_files/headers -o bin//OSDPYTHON_wrap.cxx C:/work/code/krande-pythonocc-core/src/SWIG_files/wrapper/OSD.i"
C:\work\code\krande-pythonocc-core\src\SWIG_files\wrapper\OSD.i(347) : Error: Template 'OSD_StreamBuffer' undefined.
C:\work\code\krande-pythonocc-core\src\SWIG_files\wrapper\OSD.i(348) : Error: Template 'OSD_StreamBuffer' undefined.
C:\work\code\krande-pythonocc-core\src\SWIG_files\wrapper\OSD.i(349) : Error: Template 'OSD_StreamBuffer' undefined.
[4/910] Swig compile src/SWIG_files/wrapper/NCollection.i for python
[5/910] Swig compile src/SWIG_files/wrapper/AppBlend.i for python
[6/910] Swig compile src/SWIG_files/wrapper/Message.i for python
FAILED: bin/MessagePYTHON_wrap.cxx bin/Message.py C:/work/code/krande-pythonocc-core/_build/bin/MessagePYTHON_wrap.cxx C:/work/code/krande-pythonocc-core/_build/bin/Message.py 
cmd.exe /C "cd /D C:\work\code\krande-pythonocc-core\_build && "C:\Program Files\JetBrains\CLion 2021.3\bin\cmake\win\bin\cmake.exe" -E make_directory C:/work/code/krande-pythonocc-core/_build/CMakeFiles/_Message.dir bin/ bin/ && "C:\Program Files\JetBrains\CLion 2021.3\bin\cmake\win\bin\cmake.exe" -E env SWIG_LIB=C:/ProgramData/chocolatey/lib/swig/tools/install/swigwin-4.0.2/Lib C:/ProgramData/chocolatey/bin/swig.exe -python -fvirtual -py3 -w302,401,402,412,314,509,512,504,325,503,520,350,351,383,389,394,395,404 -DCSFDB -DWIN32 -D_WINDOWS -outdir bin/ -c++ -interface _Message -IC:/work/code/krande-pythonocc-core -IC:/work/code/krande-pythonocc-core/OPENGL_INCLUDE_DIR -IC:/AibelProgs/Miniconda3/envs/pyoccdev/Library/include/opencascade -IC:/work/code/krande-pythonocc-core/src/SWIG_files/headers -o bin//MessagePYTHON_wrap.cxx C:/work/code/krande-pythonocc-core/src/SWIG_files/wrapper/Message.i"
C:\work\code\krande-pythonocc-core\src\SWIG_files\wrapper\OSD.i(347) : Error: Template 'OSD_StreamBuffer' undefined.
C:\work\code\krande-pythonocc-core\src\SWIG_files\wrapper\OSD.i(348) : Error: Template 'OSD_StreamBuffer' undefined.
C:\work\code\krande-pythonocc-core\src\SWIG_files\wrapper\OSD.i(349) : Error: Template 'OSD_StreamBuffer' undefined.
[7/910] Swig compile src/SWIG_files/wrapper/Plate.i for python
FAILED: bin/PlatePYTHON_wrap.cxx bin/Plate.py C:/work/code/krande-pythonocc-core/_build/bin/PlatePYTHON_wrap.cxx C:/work/code/krande-pythonocc-core/_build/bin/Plate.py 
cmd.exe /C "cd /D C:\work\code\krande-pythonocc-core\_build && "C:\Program Files\JetBrains\CLion 2021.3\bin\cmake\win\bin\cmake.exe" -E make_directory C:/work/code/krande-pythonocc-core/_build/CMakeFiles/_Plate.dir bin/ bin/ && "C:\Program Files\JetBrains\CLion 2021.3\bin\cmake\win\bin\cmake.exe" -E env SWIG_LIB=C:/ProgramData/chocolatey/lib/swig/tools/install/swigwin-4.0.2/Lib C:/ProgramData/chocolatey/bin/swig.exe -python -fvirtual -py3 -w302,401,402,412,314,509,512,504,325,503,520,350,351,383,389,394,395,404 -DCSFDB -DWIN32 -D_WINDOWS -outdir bin/ -c++ -interface _Plate -IC:/work/code/krande-pythonocc-core -IC:/work/code/krande-pythonocc-core/OPENGL_INCLUDE_DIR -IC:/AibelProgs/Miniconda3/envs/pyoccdev/Library/include/opencascade -IC:/work/code/krande-pythonocc-core/src/SWIG_files/headers -o bin//PlatePYTHON_wrap.cxx C:/work/code/krande-pythonocc-core/src/SWIG_files/wrapper/Plate.i"
C:\work\code\krande-pythonocc-core\src\SWIG_files\wrapper\OSD.i(347) : Error: Template 'OSD_StreamBuffer' undefined.
C:\work\code\krande-pythonocc-core\src\SWIG_files\wrapper\OSD.i(348) : Error: Template 'OSD_StreamBuffer' undefined.
C:\work\code\krande-pythonocc-core\src\SWIG_files\wrapper\OSD.i(349) : Error: Template 'OSD_StreamBuffer' undefined.
[8/910] Swig compile src/SWIG_files/wrapper/LocalAnalysis.i for python
[9/910] Building CXX object CMakeFiles\_FSD.dir\bin\FSDPYTHON_wrap.cxx.obj
[10/910] Building CXX object CMakeFiles\_Plugin.dir\bin\PluginPYTHON_wrap.cxx.obj
```

I've tried removing references to the OSD module, but that only caused more errors, so it might suggest the solution lies elsewhere.

## Further work

While I keep testing, I would appreciate any hints or feedback that would help solving this.

Best Regards
Kristoffer

Krande, 2022-04-24:
Here's an update.

By building on the latest work by trelau in his [pybinder project](https://github.com/trelau/pybinder/tree/supported_parameter_checks) I found the necessary excludes to remove the previous  errors related to `OSD_Streambuffer`. 

So now it runs a lot further before failing due to errors seemingly arbitrarily relating to `BRepAdaptor`, `Adaptor2d_OffsetCurve` and `Bnd_Box2d` depending on which platform. 

See github actions workflow logs for compilation attempts on all platforms [here](https://github.com/Krande/pythonocc-core/runs/6146885113?check_suite_focus=true) and the latest commit I made to the generate_wrapper.py script here -> https://github.com/Krande/pythonocc-generator/commit/b0f6dd7a58731aed4f3b79f9678decefe90bc1e6

Excerpt from Windows compilation attempt here

```
[111/910] Building CXX object CMakeFiles\_BRep.dir\bin\BRepPYTHON_wrap.cxx.obj
FAILED: CMakeFiles/_BRep.dir/bin/BRepPYTHON_wrap.cxx.obj 
C:\PROGRA~1\MICROS~2\2022\ENTERP~1\VC\Tools\MSVC\1416~1.270\bin\HostX64\x64\cl.exe  /nologo /TP -DCSFDB -DHAVE_CONFIG_H -DOCCT_NO_DEPRECATED -DWIN32 -DWNT -D_BRep_EXPORTS -D_WINDOWS -I%SRC_DIR% -I%SRC_DIR%\OPENGL_INCLUDE_DIR -I%PREFIX%\Lib\site-packages\numpy\core\include -I%PREFIX%\Library\include\opencascade -I%SRC_DIR%\src\SWIG_files\headers -I%SRC_DIR%\src\Tesselator -I%SRC_DIR%\src\Visualization -I%SRC_DIR%\src\MeshDataSource -I%SRC_DIR%\src\Addons -I%PREFIX%\include /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MD /O2 /Ob2 /DNDEBUG   /bigobj /wd4244 /wd4251 /wd4275 /wd4290 /showIncludes /FoCMakeFiles\_BRep.dir\bin\BRepPYTHON_wrap.cxx.obj /FdCMakeFiles\_BRep.dir\ /FS -c %SRC_DIR%\build\bin\BRepPYTHON_wrap.cxx
%SRC_DIR%\build\bin\BRepPYTHON_wrap.cxx(35867): error C2065: 'Bnd_Box2d': undeclared identifier
%SRC_DIR%\build\bin\BRepPYTHON_wrap.cxx(35867): error C2923: 'NCollection_Array1': 'Bnd_Box2d' is not a valid template type argument for parameter 'TheItemType'
%SRC_DIR%\build\bin\BRepPYTHON_wrap.cxx(35867): error C2065: 'Bnd_HArray1OfBox2d': undeclared identifier
%SRC_DIR%\build\bin\BRepPYTHON_wrap.cxx(35867): error C2059: syntax error: ')'
%SRC_DIR%\build\bin\BRepPYTHON_wrap.cxx(36377): error C2065: 'Bnd_HArray1OfSphere': undeclared identifier
%SRC_DIR%\build\bin\BRepPYTHON_wrap.cxx(36377): error C2059: syntax error: ')'
%SRC_DIR%\build\bin\BRepPYTHON_wrap.cxx(36542): error C2065: 'Bnd_HArray1OfBox': undeclared identifier
%SRC_DIR%\build\bin\BRepPYTHON_wrap.cxx(36542): error C2059: syntax error: ')'
%SRC_DIR%\build\bin\BRepPYTHON_wrap.cxx(36563): error C2065: 'Bnd_HArray1OfBox2d': undeclared identifier
%SRC_DIR%\build\bin\BRepPYTHON_wrap.cxx(36563): error C2059: syntax error: ')'
%SRC_DIR%\build\bin\BRepPYTHON_wrap.cxx(36617): error C2065: 'Bnd_Sphere': undeclared identifier
%SRC_DIR%\build\bin\BRepPYTHON_wrap.cxx(36617): error C2923: 'NCollection_Array1': 'Bnd_Sphere' is not a valid template type argument for parameter 'TheItemType'
%SRC_DIR%\build\bin\BRepPYTHON_wrap.cxx(36617): error C2065: 'Bnd_HArray1OfSphere': undeclared identifier
%SRC_DIR%\build\bin\BRepPYTHON_wrap.cxx(36617): error C2059: syntax error: ')'
%SRC_DIR%\build\bin\BRepPYTHON_wrap.cxx(36620): error C2065: 'Bnd_HArray1OfBox': undeclared identifier
%SRC_DIR%\build\bin\BRepPYTHON_wrap.cxx(36620): error C2059: syntax error: ')'
[112/910] Building CXX object CMakeFiles\_BRepTools.dir\bin\BRepToolsPYTHON_wrap.cxx.obj
[113/910] Swig compile src/SWIG_files/wrapper/BRepAdaptor.i for python
ninja: build stopped: subcommand failed.
    sys.exit(main())
  File "C:\Miniconda\envs\condabuild\lib\site-packages\conda_build\cli\main_build.py", line 488, in main
    execute(sys.argv[1:])
  File "C:\Miniconda\envs\condabuild\lib\site-packages\conda_build\cli\main_build.py", line 477, in execute
    outputs = api.build(args.recipe, post=args.post, test_run_post=args.test_run_post,
  File "C:\Miniconda\envs\condabuild\lib\site-packages\conda_build\api.py", line 186, in build
    return build_tree(
  File "C:\Miniconda\envs\condabuild\lib\site-packages\conda_build\build.py", line 3088, in build_tree
    packages_from_this = build(metadata, stats,
  File "C:\Miniconda\envs\condabuild\lib\site-packages\conda_build\build.py", line 2179, in build
    windows.build(m, build_file, stats=build_stats, provision_only=provision_only)
  File "C:\Miniconda\envs\condabuild\lib\site-packages\conda_build\windows.py", line 297, in build
    check_call_env(cmd, cwd=m.config.work_dir, stats=stats, rewrite_stdout_env=rewrite_env)
  File "C:\Miniconda\envs\condabuild\lib\site-packages\conda_build\utils.py", line 410, in check_call_env
    return _func_defaulting_env_to_os_environ('call', *popenargs, **kwargs)
  File "C:\Miniconda\envs\condabuild\lib\site-packages\conda_build\utils.py", line 390, in _func_defaulting_env_to_os_environ
    raise subprocess.CalledProcessError(proc.returncode, _args)
subprocess.CalledProcessError: Command '['cmd.exe', '/d', '/c', 'conda_build.bat']' returned non-zero exit status 1.
Error: Process completed with exit code 1.
```

Krande, 2022-08-10:
Hey, I took another stab at making modifications for the generator to work with OCCT v7.6, and have now gotten a bit further. 

Now I am reaching almost 50% of the compilation of pythonocc-core for occt v7.6.2 by basically of adding/removing headers and skipping various function members etc..

However, I am running into an issue that I am unable to solve using tricks that seems to have worked until now. 

```
...
[411/910] Building CXX object CMakeFiles\_HLRTopoBRep.dir\bin\HLRTopoBRepPYTHON_wrap.cxx.obj
FAILED: CMakeFiles/_HLRTopoBRep.dir/bin/HLRTopoBRepPYTHON_wrap.cxx.obj 
C:\PROGRA~1\MICROS~2\2022\ENTERP~1\VC\Tools\MSVC\1416~1.270\bin\HostX64\x64\cl.exe  /nologo /TP -DCSFDB -DHAVE_CONFIG_H -DHAVE_FREEIMAGE -DHAVE_FREETYPE -DHAVE_OPENGL_EXT -DHAVE_RAPIDJSON -DHAVE_VTK -DOCCT_NO_DEPRECATED -DUNICODE -DVTK_OPENGL2_BACKEND -DWIN32 -DWNT -D_CRT_NONSTDC_NO_DEPRECATE -D_CRT_SECURE_NO_WARNINGS -D_HLRTopoBRep_EXPORTS -D_UNICODE -D_WINDOWS -I%SRC_DIR% -I%SRC_DIR%\OPENGL_INCLUDE_DIR -I%PREFIX%\Lib\site-packages\numpy\core\include -I%PREFIX%\Library\include -I%PREFIX%\Library\include\opencascade -I%SRC_DIR%\src\SWIG_files\headers -I%SRC_DIR%\src\Tesselator -I%SRC_DIR%\src\Visualization -I%SRC_DIR%\src\MeshDataSource -I%SRC_DIR%\src\Addons -I%PREFIX%\include /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MD /O2 /Ob2 /DNDEBUG   /bigobj /wd4244 /wd4251 /wd4275 /wd4290 /showIncludes /FoCMakeFiles\_HLRTopoBRep.dir\bin\HLRTopoBRepPYTHON_wrap.cxx.obj /FdCMakeFiles\_HLRTopoBRep.dir\ /FS -c %SRC_DIR%\build\bin\HLRTopoBRepPYTHON_wrap.cxx
%PREFIX%\Library\include\opencascade\Contap_ArcFunction.hxx(47): error C2065: 'Adaptor2d_Curve2d': undeclared identifier
%PREFIX%\Library\include\opencascade\Contap_ArcFunction.hxx(47): error C2923: 'opencascade::handle': 'Adaptor2d_Curve2d' is not a valid template type argument for parameter 'T'
%PREFIX%\Library\include\opencascade\Contap_ArcFunction.hxx(81): error C2065: 'Adaptor2d_Curve2d': undeclared identifier
%PREFIX%\Library\include\opencascade\Contap_ArcFunction.hxx(81): error C2923: 'opencascade::handle': 'Adaptor2d_Curve2d' is not a valid template type argument for parameter 'T'
%PREFIX%\Library\include\opencascade\Contap_ArcFunction.lxx(45): error C2065: 'Adaptor2d_Curve2d': undeclared identifier
%PREFIX%\Library\include\opencascade\Contap_ArcFunction.lxx(45): error C2923: 'opencascade::handle': 'Adaptor2d_Curve2d' is not a valid template type argument for parameter 'T'
%PREFIX%\Library\include\opencascade\Contap_ArcFunction.lxx(47): error C2678: binary '=': no operator found which takes a left-hand operand of type 'opencascade::handle' (or there is no acceptable conversion)
...
```

The full compilation output can be found at:
https://github.com/Krande/pythonocc-core/runs/7748146537?check_suite_focus=true#step:5:1042

While the files are at:
https://github.com/Krande/pythonocc-core/tree/v7.6.2a1

I have tried to include `Adaptor2d_Curve2d.hxx` and/or `Adaptor2d.hxx` to the `Contap.i` with no luck. 

The only thing I've tried that seems to work is to go into the source `Contap_ArcFunction.hxx` located in the `opencascade` include directory in the conda package and manually add `Adaptor2d_Curve2d.hxx` to the include statements. 

```cpp
#ifndef _Contap_ArcFunction_HeaderFile
#define _Contap_ArcFunction_HeaderFile

#include <Adaptor3d_Surface.hxx>
#include <Adaptor2d_Curve2d.hxx>
#include <Contap_TFunction.hxx>
#include <gp_Dir.hxx>
#include <gp_Pnt.hxx>
#include <TColgp_SequenceOfPnt.hxx>
#include <IntSurf_Quadric.hxx>
#include <math_FunctionWithDerivative.hxx>

class Contap_ArcFunction  : public math_FunctionWithDerivative
```

But as far as I can tell this would require a custom compiled OCCT package on conda which seems unpractical to say the least. 

@tpaviot @aothms Any advice you could give on this would be much appreciated?

aothms, 2022-08-10:
It seems this has been fixed at some point in OCCT upstream where the header is included https://github.com/Open-Cascade-SAS/OCCT/commit/7109a4ae5843002ea7a35ed2b5aafba7ccef13d4#diff-89e4ed09db4b90a88f132a00171a3172f26279c2269e97c6861f90f75c595246

Didn't take the time to figure out what tags include this commit. Maybe 7.6.3 🙏 ?

Krande, 2022-08-10:
@aothms Thank you for the quick response! As a matter of fact I just now figured out what was wrong in my attempt at solving this. 

I was adding the headers in the wrong file `Contap.i` instead of in `HLRTopoBRep.i` (which is what the error actually was telling me). 

I just missed completed where it actually failed:  `FAILED: CMakeFiles/_HLRTopoBRep.dir/bin/HLRTopoBRepPYTHON_wrap.cxx.obj` 

So the tricks that worked before seems to still work. I'll let you know once I have this at a 100% :)

aothms, 2022-08-10:
Great, still a bit of a bug though that they resolved, headers should include the headers they depend on.

Krande, 2022-08-10:
Yes, I see there is a 7.6.3 Tag, but I'll run through all of it using 7.6.2 to find other issues first. 

Here's another issue. 

There is a specific pythonocc-core class `ShapeTesselator` that seems to point to `Nodes()` and `UVNodes()` methods on Poly_Triangulation that are no longer available in 7.6: 

__Poly_Triangulation does no more provide access to internal array structures: methods Nodes(), ChangeNode(), Triangles(), ChangeTriangle(), UVNodes(), ChangeUVNode(), Normals() have been removed. Methods of [Poly_Triangulation](https://dev.opencascade.org/doc/refman/html/class_poly___triangulation.html) for accessing individual nodal properties / triangles by index and implementing copy semantics should be used instead. The same is applicable to [Poly_PolygonOnTriangulation](https://dev.opencascade.org/doc/refman/html/class_poly___polygon_on_triangulation.html) interface.__

https://dev.opencascade.org/doc/overview/html/occt__upgrade.html#upgrade_occt760_poly

```cpp
Handle(Poly_Triangulation) myT = BRep_Tool::Triangulation(myFace, aLocation);

if (myT.IsNull()) {
    invalidFaceTriCount++;
    continue;
}

aface *this_face = new aface;

//write vertex buffer
const TColgp_Array1OfPnt& Nodes = myT->Nodes();
this_face->vertex_coord = new double[Nodes.Length() * 3];
this_face->number_of_coords = Nodes.Length();
for (Standard_Integer i = Nodes.Lower(); i <= Nodes.Upper(); i++) {
  gp_Pnt p = Nodes(i).Transformed(aLocation.Transformation());
  this_face->vertex_coord[((i-1) * 3)+ 0] = p.X();
  this_face->vertex_coord[((i-1) * 3)+ 1] = p.Y();
  this_face->vertex_coord[((i-1) * 3)+ 2] = p.Z();
}
```
https://github.com/Krande/pythonocc-core/blob/v7.6.2a1/src/Tesselator/ShapeTesselator.cpp#L109-L128

I am guessing this is just a matter of basically rewriting parts of the script to no longer point to the `Nodes()` method by substituting `Nodes.Length()` with `myT.NbNodes()`

Any hints on how I should rewrite `Standard_Integer i = Nodes.Lower(); i <= Nodes.Upper(); i++`?

aothms, 2022-08-10:
I think you can follow what IfcOpenShell does

https://github.com/IfcOpenShell/IfcOpenShell/blob/v0.7.0/src/ifcgeom_schema_agnostic/IfcGeomRepresentation.cpp#L154-L156

~~~C++
auto p = myT->Node(i).Transformed(loc).XYZ();
this_face->vertex_coord[((i-1) * 3)+ 0] = p.X();
this_face->vertex_coord[((i-1) * 3)+ 1] = p.Y();
this_face->vertex_coord[((i-1) * 3)+ 2] = p.Z();
~~~

(untested)

Krande, 2022-08-10:
Thanks for the tip! 

But I think I might have found a workaround with limited impact on the existing code base using

```cpp
const TColgp_Array1OfPnt& Nodes = myT->MapNodeArray()->Array1();
```

I guess I'll find out eventually if my hacks and workarounds does/does not break things!

Krande, 2022-08-11:
A new day, a little progress and another issue :) 

I ran into an issue with wrapping an `enum` today when compiling the `RWGltf` module.

```
FAILED: CMakeFiles/_RWGltf.dir/bin/RWGltfPYTHON_wrap.cxx.obj 
C:\PROGRA~1\MIB055~1\2022\PROFES~1\VC\Tools\MSVC\1431~1.311\bin\Hostx86\x64\cl.exe  /nologo /TP -DCSFDB -DHAVE_CONFIG_H -DHAVE_FREEIMAGE -DHAVE_FREETYPE -DHAVE_OPENGL_EXT -DHAVE_RAPIDJSON -DHAVE_VTK -DOCCT_NO_DEPRECATED -DUNICODE -DVTK_OPENGL2_BACKEND -DWIN32 -DWNT -D_CRT_NONSTDC_NO_DEPRECATE -D_CRT_SECURE_NO_WARNINGS -D_RWGltf_EXPORTS -D_UNICODE -D_WINDOWS -IC:\Users\ofskrand\CLionProjects\krande-pythonocc-core -IC:\Users\ofskrand\CLionProjects\krande-pythonocc-core\OPENGL_INCLUDE_DIR -IC:\AibelProgs\Miniconda3\envs\pyoccdev\Lib\site-packages\numpy\core\include -IC:\AibelProgs\Miniconda3\envs\pyoccdev\Library\include -IC:\AibelProgs\Miniconda3\envs\pyoccdev\Library\include\opencascade -IC:\Users\ofskrand\CLionProjects\krande-pythonocc-core\src\SWIG_files\headers -IC:\Users\ofskrand\CLionProjects\krande-pythonocc-core\src\Tesselator -IC:\Users\ofskrand\CLionProjects\krande-pythonocc-core\src\Visualization -IC:\Users\ofskrand\CLionProjects\krande-pythonocc-core\src\MeshDataSource -IC:\Users\ofskrand\CLionProjects\krande-pythonocc-core\src\Addons -external:IC:\AibelProgs\Miniconda3\envs\pyoccdev\include -external:W0 /DWIN32 /D_WINDOWS /W3 /GR /EHsc /MD /O2 /Ob2 /DNDEBUG   /bigobj /wd4244 /wd4251 /wd4275 /wd4290 -std:c++14 /showIncludes /FoCMakeFiles\_RWGltf.dir\bin\RWGltfPYTHON_wrap.cxx.obj /FdCMakeFiles\_RWGltf.dir\ /FS -c ..\bin\RWGltfPYTHON_wrap.cxx
..\RWGltfPYTHON_wrap.cxx(22973): error C2065: 'ShapeMapGroup_Nodes': undeclared identifier
..\RWGltfPYTHON_wrap.cxx(22973): error C2660: 'SWIG_Python_SetConstant': function does not take 2 arguments
..\RWGltfPYTHON_wrap.cxx(1178): note: see declaration of 'SWIG_Python_SetConstant'
..\RWGltfPYTHON_wrap.cxx(22974): error C2065: 'ShapeMapGroup_Meshes': undeclared identifier
..\RWGltfPYTHON_wrap.cxx(22974): error C2660: 'SWIG_Python_SetConstant': function does not take 2 arguments
..\RWGltfPYTHON_wrap.cxx(1178): note: see declaration of 'SWIG_Python_SetConstant'
..\RWGltfPYTHON_wrap.cxx(22975): error C2065: 'ShapeMapGroup_PrimArray': undeclared identifier
..\RWGltfPYTHON_wrap.cxx(22975): error C2660: 'SWIG_Python_SetConstant': function does not take 2 arguments
..\RWGltfPYTHON_wrap.cxx(1178): note: see declaration of 'SWIG_Python_SetConstant'
```

The offending enum definitions in [RWGltf.i](https://github.com/Krande/pythonocc-core/blob/v7.6.2a1/src/SWIG_files/wrapper/RWGltf.i):

```
...
enum ShapeMapGroup {
	ShapeMapGroup_Nodes = 0,
	ShapeMapGroup_Meshes = 1,
	ShapeMapGroup_PrimArray = 2,
};
...
class ShapeMapGroup(IntEnum):
	ShapeMapGroup_Nodes = 0
	ShapeMapGroup_Meshes = 1
	ShapeMapGroup_PrimArray = 2
ShapeMapGroup_Nodes = ShapeMapGroup.ShapeMapGroup_Nodes
ShapeMapGroup_Meshes = ShapeMapGroup.ShapeMapGroup_Meshes
ShapeMapGroup_PrimArray = ShapeMapGroup.ShapeMapGroup_PrimArray
...
```

And definitions in [RWGltf.pyi](https://github.com/Krande/pythonocc-core/blob/v7.6.2a1/src/SWIG_files/wrapper/RWGltf.pyi):

```
class ShapeMapGroup(IntEnum):
    ShapeMapGroup_Nodes: int = ...
    ShapeMapGroup_Meshes: int = ...
    ShapeMapGroup_PrimArray: int = ...

ShapeMapGroup_Nodes = ShapeMapGroup.ShapeMapGroup_Nodes
ShapeMapGroup_Meshes = ShapeMapGroup.ShapeMapGroup_Meshes
ShapeMapGroup_PrimArray = ShapeMapGroup.ShapeMapGroup_PrimArray
```

Krande, 2022-08-11:
Okay, I am happy to report that pythonocc-core now compiles successfully on all 3 platforms! 

In fact, to my surprise all tests are passing on `windows`. The excitement was short lived however. It seems bld.bat and build.sh are not testing the same things on my branch. 

Both Linux and OSX do pass a lot of tests, but are failing somewhere in the `core_geometry_unittest.TestGeometry` suite.

[image]

Here are the full logs from the github actions pipelines: 

https://github.com/Krande/pythonocc-core/runs/7787358841?check_suite_focus=true

The conda package of pythonocc-core v7.6.2 is available for testing on windows here -> https://anaconda.org/krande/pythonocc-core/files. 

I will do some testing myself as well and see if the failing tests on linux are due to changes in the occt v7.6 api or in my SWIG adaption.

@looooo I thought you might be interested in this given you mentioned in https://github.com/tpaviot/pythonocc-core/issues/1111 an interest in updating pythonocc-core to v7.6.

aothms, 2022-08-13:
Wonderful progress!

tpaviot, 2022-08-14:
Thank you @Krande for your work, and sorry for being off for a few months after I took a new job position. I updated the generator to deal with occt7.6.2 headers. Just a few words regarding your commits: there's no need to create a specific swig_modifier for each occt version. Indeed related changes may be needed in future versions, and adding new headers on top of swig files does not create any backward/forward incompatibility. Note that the generate_wrapper.py script is formatted with black using default settings. The ShapeTesselator.cpp file needed a couple of additional tweaks to run on my linux machine and avoir segfaults. I also had to disable the test_curve_adaptor test, which is related to a weird issue, I'm not sure it is even needed. A build is running on Azure, I will report the status.

Anyway, congrats for your work! It's not obvious to enter the pythonocc generator, I'm glad to welcome a new contributor. Feel free to submit pull requests, splitted into atomic commits that do solve a documented issue (the build process using conda/azure is quite unstable and may suffer from many side effects).

tpaviot, 2022-08-14:
and thank you @aothms for your pointer to the ifcopenshell cpp code

tpaviot, 2022-08-14:
@Krande the next step is to check/update the pythonocc-demos to the new version.

Krande, 2022-08-14:
@tpaviot Just happy to help, and thank you for the feedback! 

Regarding the changes I made to the `generate_wrapper.py`. I agree with your comments. My changes was primarily intended to increase my own familiarity with the SWIG generator process and was never intended to replace the current code base.

tpaviot, 2022-08-15:
changes for occt762 have been merged into master, pythonocc-core 7.6.2 compilation and tests work fine

Krande, 2022-08-15:
That's great!

Then I think we can close this issue.

closed 2022-08-15 by Krande