tpaviot/pythonocc-generator #90: C# Generator

cesarecaoduro, 2024-02-27:
Hi, I’m not familiar with SWIG and I’m trying really hard to understand how to create a configuration file to wrap OCCT for c#.
Would this python generator, able to generate code for c# as well?
Thanks in advance and thanks for all the hard work you’ve invested for the community.

tpaviot, 2024-02-29:
swig is able to generate c# wrappers. There are many python specific customizations in pythonocc-generator, however it should be possible to use most of the interface files to generate c#, however I have not experimented c# wrapping using swig

cesarecaoduro, 2024-02-29:
Thanks for answering  my question. Unfortunately I have zero experience in
this space, hence I was asking.
Not sure if it is of your interest to build a C# wrapper, but I can assure
you that would be really well received by the community.
Cheers,
Cesare